ni/niveristand-scan-engine-ethercat-custom-device #9: Scan Engine and EthercaT custom device interface in veristand

anike94, 2018-12-19:
Hello
I am currently working on a hybrid mode interface with veristand. By going through forums we come to know it requires scan engine and EtherCAT custom device so I downloaded the
  NI VeriStand 2017 with NI-RIO 17.0+ and EtherCAT 17.0+:
   Scan Engine and EtherCAT 2017  
After I downloaded I directly placed the scan engine folder in \Documents\National Instruments\NI VeriStand 2017\Custom Devices \. I am able to add the custom device in veristand (Configuring modules and bit file) but when I try to deploy in Veristand RT it shows an error(Please find the attached image 2 states that VI is not executable), So I try to open the source code it shows an error(Please find the attached image 3 states that LabVIEW error code 3)
Can anyone please guide me to know exactly what is the problem and also please share the working custom device for 2017 version
I am using LabVIEW and veristand 2017 and cRIO-9076 also I will attach all the softwares installed
[image]
[image]
[image]
[image]
[image]
[image]

tannerblair, 2018-12-19:
@christinesparks, can you offer any insight here?

christinesparks, 2018-12-20:
So a couple things to note here:

1. The cRIO chassis you are trying to use is an older chassis that only has a single ethernet port which makes it incompatible with EtherCAT (which you already knew). This is also a controller with limited memory and processor capability and you may struggle to get VeriStand running on it successfully even if we do resolve this issue.
2. You are getting the LV error "error code 3: could not load front panel" because you are trying to open VIs in the build LLBs. Opening them will likely not resolve the deployment issue you are facing, but if you wanted to do so you'd want to download the source from this GitHub page instead of using the built LLBs.

I believe the source introduced a necessary dependency on the ethercat driver at some point during its development. I am planning on testing this theory and then will file an additional issue so that NI can investigate the issue. There is a small chance of this being fixed immediately so if you are under tight timelines I would suggest either getting a hold of a different cRIO (I'd recommend the 906x or 903x for VeriStand) or by using exclusively FPGA I/O and utilizing the native FPGA support or looking at the FPGA Custom Device (https://github.com/NIVeriStandAdd-Ons/FPGA-Custom-Device).

anike94, 2018-12-23:
Hello
Thank you for your reply. Is it possible for you to share the working Scan Engine and EthercaT custom device that you have tested for LabVIEW and Veristand 2017 Version, because I  downloaded from online may be it get corrupted. So it will be great help if you share the custom device
Thank You

rtzoeller, 2019-02-22:
@anike94 were you able to work with NI support to resolve or work around the issues you were experiencing? Can this issue be closed?

I will file another issue to track the behavior that using the custom device with exclusively local modules still requires the EtherCAT driver to be installed.

rtzoeller, 2019-03-22:
I've captured the underlying bug [here](https://github.com/ni/niveristand-scan-engine-ethercat-custom-device/issues/26), closing this.

closed 2019-03-22 by rtzoeller