ni/niveristand-scan-engine-ethercat-custom-device #15: NI-9049, local chassis modules not detected in some scenarios.

VincentRuault, 2019-01-30:
**Describe the bug**
When dealing with some configurations (HW and SW), some modules (local chassis modules in my case) are not detected correctly by the Auto-detect features of the SEEC CD. I did the latest test with SEEC CD 4.4.2 with NI-VS 2018. I needed it because of Issue #12.

HW configuration used:
- NI-9049 or NI-9040 and following C-Series modules in local chassis: NI-9263, NI-9225 and NI-9265
- NI-9145 EtherCAT chassis: NI-9239, NI-9263, NI-9274

When auto-detect doesn't work, deploy one time, undeploy and redo an auto-deploy and it works.
Symptoms are really similar to what is described in closed CAR  #173866 (fixed).

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new Sysdef and add cRIO-904x chassis and configure it in FPGA mode. 
2. Add SEEC CD, perform an auto-detect, if not already, local chassis slots should be changed (if applicable) to FPGA (check in MAX, sometimes a click on slot is required): EtherCAT chassis and its modules are detected correctly. It's expected behavior.
3. Deploy
4. Undeploy
5. Reopen Sysdef, change Target mode to Scan mode.
6. Under SEEC CD, peform an auto-detect, local chassis slots should be changed automatically to Scan Mode (check in MAX, sometimes a click on slot is required): EtherCAT chassis and its modules are detected correctly but not the modules from local chassis. I checked with debug code and variables are created correctly but there is a 65704 error happening in sub-VI "Host - Get Local and ECAT Masters.vi".
7. Deploy Sysdef
8. Undeploy Sysdef
9. Do a new Auto-Detect and, now, local chassis modules are also listed.

**Expected behavior**
Modules from local chassis should be detected.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. Pharlap]
 - Version [e.g. 22]

**Additional context**
I had another customer using LabVIEW-RT with same problem on a different cRIO. This customer had stolen some code from SEEC CD source to implement dynamic detection of configuration. In his case, missing modules were in some EtherCAT remote chassis but same error (65704 was happening). I could never reproduced it.

VincentRuault, 2019-02-12:
Extra information. To reproduce the issue, the NI-9145 is not required. I redid the test without EhterCAT chassis. In main local chassis, here are the modules configuration and SW versions in the NI-9049 chassis.
[image]

VincentRuault, 2019-02-13:
Below is a potential Workaround while the bug is fixed. The idea is to detect the 65704 error occurrence and if it happens, let the user know (via a dialogue) that he needs to deploy a first time and perform another auto-detect to complete the configuration detection. Really not ideal but, at least, he will bring user some guidance.
[image]
NOTE: I tried to flatten to string the RSI URL and then add it to the Array "All Local Mods" of URL for Local Modules. But, then, in following VI: "Host - Get Module Data.vi", same error 65705 happens again.

This behavior really looks like the closed CAR: #173866.

buckd, 2019-02-13:
@VincentRuault From your screenshot, obviously the items exist in `Children[]`. However, it's most likely that since they're not deployed, they aren't of one of the three types this VI is checking for. Can you drop a property node for each child and get the **Class Identifier**? That may tell us why the modules aren't being detected. You might then be able to add that class to do something to add that class to your list of modules.

VincentRuault, 2019-02-14:
Hi @buckd ,
Thank-you for your answer. Please check the capture below. I think I tried what you described but call to property ClassIdentifier? returns same 65704 error. Can you confirm it's what you were asking?
Can you have a look at probes 2 and 3 in capture?

[image]

buckd, 2019-02-14:
>I think I tried what you described but call to property ClassIdentifier? returns same 65704 error. Can you confirm it's what you were asking?

That's what I was wanting to see, but I was hoping for something besides that error. It looks like the module names are the defaults, so they should be able to resolve the URL. I agree this looks very similar to the CAR you mentioned, which is a driver CAR and not something we can fix in the custom device.

I would suggest talking to the driver team and to see if they can look into why this is happening and if there is anything we can do to resolve this in the custom device. We will keep this issue open, but we likely won't pull in any workarounds without having a more complete understanding of the underlying issue.  We will also soon be looking at all the GitHub issues and prioritizing them with our other work, so we will look at this, but I can't promise when. If the driver team has an understanding and recommendation, we are much more likely to follow their guidance.

VincentRuault, 2019-02-14:
Thank-you.
Which driver is-it? Is-it NI-RIO?

VincentRuault, 2019-04-26:
Hi,

@buckd and @rtzoeller , CAR #729217 has been updated by NI-SH team.
Can you have a look at it and let me know what you think and what we can do?

About rationales:

- I understand we are de-investing in RSI but, if the case, what is the replacement? Could that replacement be used for SEEC CD instead, if it makes sense?

- I'm not sure the #4 can be seen as a workaround. We should put ourselves in customer's shoes, in terms of user-experience, with SEEC CD: how many dialogues, are-we going to show in the end?

Regards

Vincent

VincentRuault, 2019-05-06:
Addressed in Pull Request #38

closed 2019-05-06 by VincentRuault