ni/niveristand-scan-engine-ethercat-custom-device #18: Support the 9266

epbray1, 2019-02-15:
**Is your feature request related to a problem? Please describe.**
Want to upgrade to 2018 and use the NI-supported SEECD. Customer project needs the 9266, and they are currently using a modified-version of the SEECD in 2017.

**Describe the solution you'd like**
Add support for the 9266.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

rtzoeller, 2019-02-19:
Thanks for filing an issue!

The 9266 looks almost identical to the 9265, which means adding support for this to https://github.com/ni/niveristand-scan-engine-module-libraries should be relatively straightforward.

rtzoeller, 2019-02-25:
I have modified the custom device to support the 9266, and confirmed that the module works as expected in both local and EtherCAT chassis.

Unfortunately, I have been having issues with this change breaking compatability with existing system definitions. I don't have any more time to commit to this right now, and won't be merging the changes until we fix the backwards compatibility issues.

If you need a built version of the custom device that supports the 9266 immediately, please let me know and I can work with you.

jashnani, 2019-04-30:
Fixed in https://github.com/ni/niveristand-scan-engine-module-libraries/pull/10

closed 2019-04-30 by jashnani