ni/niveristand-scan-engine-ethercat-custom-device #19: Bug in DBL and SGL support for 3rd party EtherCAT slaves

keprt, 2019-02-19:
**Description**
If the 3rd party slaves contains DBL or SGL scan engine variable type, SEECD fails on deploy.

**Steps to reproduce the behavior**
Add 3rd party slave, that has DBL or SGL datatype. Autodetect modules. Deploy

**Expected behavior**
System should be able to read and write DBL and SGL variables properly.

**Screenshots**
Printscreens from **fixed** version:
UDV Data Type.ctl needs to get DBL and SGL item
[image]

TypeCodeToEnum needs Case A and 9
[image]

Write and Read needs SGL and DBL case
[image]

[image]

**Desktop (please complete the following information):**
 - OS: all RT

**Additional context**
I've already implemented a fix for customer based on older version of SEECD. So you could contact me and we could do a merge together.

rtzoeller, 2019-02-19:
@keprt can you list some examples of devices which are broken by this? It would be nice to have for validation.

keprt, 2019-02-19:
@rtzoeller probably don't want to list all here. I will send you email

fmussi, 2019-10-30:
I don't know if this is the right place, but with the 4.5 version that we ship with VS 2019R2, I cannot read SGL variables. I always get 0 values where LV or Distributed system manager returns the correct SGL value. Is it a known issue?

rtzoeller, 2019-10-30:
@fmussi that is the expected behavior; this issue is still open.

This is not an issue we are actively prioritizing in the backlog, however we would welcome a pull request adding support if you are able to validate the functionality.

closed 2020-01-20 by rtzoeller