ni/niveristand-scan-engine-ethercat-custom-device #47: NI 9237 [Dynamic] Shunt Cal Functionality / Logic

tarmanan, 2019-06-18:
I am using Veristand 2017 in FPGA mode with a 9081 controller and 9144 EtherCAT chassis.  The 9237 is located in the 9144 EtherCAT chassis.  

I'm looking to understand how the shunt calibration functions.   If I select the checkbox for "Shunt Calibration Enabled", what should happen? If I select "Allow Dynamic Shunt Calibrations", I receive additional boolean options under "Module Configuration".  I would have expected, after deploying my project, that if "AI0 Shunt Calibration" is set to "1", then I would be performing a shunt calibration.  

I am using a Honeywell Unamplified Pressure transducer that will go to 80% of full scale when shunted.  [344.7 bar pressure transducer should read 276 bar when shunted]  

My pressure transducers are reading properly and I've checked them with a pressure calibrator.  However, nothing happens when I change "AI0 Shunt Calibration" from "1" [default] to "0".

[image]

tarmanan, 2019-06-18:
[image]

rtzoeller, 2019-06-18:
Hi @tarmanan,

We haven't had a chance to look at this in depth, but from a cursory glance at the source code I believe I've found the issue. I'll track down an NI-9237 to validate the fix with, but that may not happen immediately.

Module reconfiguration (e.g. modifying `AI0 Shunt Calibration`) requires taking the EtherCAT slave (in this case the NI-9144) out of active mode and into configuration mode. Since this is disruptive for EtherCAT modules, it does not happen automatically and instead requires the user to issue a command by setting the `Reconfigure EtherCAT Modules` channel.

I believe setting the `Reconfigure EtherCAT Modules` channel to a non-zero value will force the reconfiguration to occur as you are expecting.

rtzoeller, 2019-06-19:
I was able to test an NI-9237 in both a NI-9144 and NI-9145 chassis and reproduced the issue you have been having.

I can confirm that setting `Reconfigure EtherCAT Modules` to 1 as described above should cause the EtherCAT chassis to be reconfigured and the current value of `AI0 Shunt Calibration` to take effect.

I've filed issue https://github.com/ni/niveristand-scan-engine-ethercat-custom-device/issues/48 and https://github.com/ni/niveristand-scan-engine-ethercat-custom-device/issues/49 to capture the lack of documentation/descriptions on the channels and sections you have been interacting with.

@tarmanan can you confirm this resolves your question?

tarmanan, 2019-06-19:
Thank you very much, @rtzoeller .  

When you set  Reconfigure EtherCAT Modules to "1", what is the expected output?  If Reconfigure EtherCAT Modules=1, does the 9237 physically shunt, for example, AI0 input terminal EX- with AI-, using a 100kohm shunt?  When does this happen - when the project is deployed?  

[http://www.ni.com/pdf/manuals/374186a_02.pdf, page 11, Shunt Calibration Resistance = 100 kohm]

tarmanan, 2019-06-19:
Additional reference:  https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z000000PA73SAG&l=en-US

rtzoeller, 2019-06-19:
> When you set Reconfigure EtherCAT Modules to "1", what is the expected output? ... When does this happen - when the project is deployed?

`Reconfigure EtherCAT Modules` provides a way to update the hardware configuration to match the configuration dynamically defined by the channels in the `Module Configuration` section. The module configuration is applied to the hardware initially on deployment, and then again whenever it is refreshed by the above process.

For example, configuring the NI-9237 like this 
[image]
will result in the `AI0 Shunt Calibration` channel having a value of 1 on deployment, and the other `AI* Shunt Calibration` channels having a value of 0. The hardware will be in sync with this state initially.

If you then update `AI1 Shunt Calibration` to be 1 and `AI0 Shunt Calibration` to be 0, the hardware will remain in its original configuration until `Reconfigure EtherCAT Modules` is set to 1. 

---

I am not familiar with shunting the NI-9237 specifically; I recommend going through the standard NI support channels to find out more information about the module. We are controlling it through the IO Variable interface in Scan Mode.
[image]

tarmanan, 2019-07-16:
@rtzoeller , thank you for your assistance and clarification.  In years past, a cable from Honeywell [RJ50 to 6-pin Bendix, 060-0604-197-01] was listed as an accessory for the 9237, either through Honeywell or NI.  I purchased several from Honeywell for our test fixtures.  After going through this exercise, I realized that RJ50 pins 1 and 10 [Shunt Cal] are not connected on the 6-pin Bendix end of the harness.  Otherwise, the cables interface properly between the 9237 and Honeywell transducers.
Thanks again for your help.  I now know what to expect, from an operator interface, when 'Shunt Calibration is Enabled'.

closed 2019-07-16 by tarmanan