ni/niveristand-scan-engine-ethercat-custom-device #52: Bitfile for XNET cModule in cRIO-906x Chassis

tarmanan, 2019-07-19:
**Please describe the problem you're facing.**
Confusion regarding the requirement for a bitfile to support XNET in cRIO chassis, using Scan Engine for EtherCAT with VeriStand 2017.  On previous projects, I have used a bitfile required for cRIO-908x when Scan Engine for EtherCAT is used with XNET [9862] module in local chassis.  On this project, I am using a cRIO-9068.  Ultimately, my challenge is using a specialty digital module [ie. 9411] in the local chassis with an XNET module [9862].  My hope was to use the native Specialty Digital Settings for the 9411 in the local chassis rather than making an FPGA with UDV's to support the 9411 + 9862 together in the local chassis.  

**Describe the expected behavior**
Understanding requirements.  Refer to document https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z000000kKTtSAM&l=en-US

**Describe troubleshooting steps you've taken.**
1)  With cRIO-**9068**, followed https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z000000kKTtSAM&l=en-US [Does this not apply for projects using the Custom Device "Scan Engine and EtherCAT"?
2)  With cRIO-**9068**, tried deploying project in all three "FPGA / Scan Interface Mode Options":  Use Current; Scan Mode; and FPGA Mode
3)  With cRIO-**_9081_**, tried creating a blank bitfile for the 9862 in slot 1 and 9411 in slot 2, with remaining slots for analog [NI 9201].  

**Additional context**
Is it possible for the XNET and Specialty Digital Modules to play nicely on the same local chassis, without requiring an FPGA with UDV's for the Specialty Digital Module?  
[image]

Karl-G1, 2019-07-19:
@tarmanan Unfortunately, this is a known issue with the Scan Engine and EtherCAT Custom Device. In your cRIO-9068, Specialty Digital configurations are not compatible with XNET C Series modules in the same controller. This affects all targets except for the latest CompactRIO with DAQmx controllers (cRIO-904x and 905x). The three workarounds are to:
- use the FPGA with UDVs (as you mentioned)
- use the Specialty Digital modules in an EtherCAT Expansion Chassis (NI 9145)
- use a cRIO-904x or 905x controller

We have investigated workarounds in the past, and the changes needed are at the driver level. Because of the complexity of the changes and the fact that newer controllers do not have the same issue, this bug is unlikely to be fixed for older controllers.

tarmanan, 2019-07-19:
@Karl-G1 thank you for your follow-up.  I'm pleased to know that the 904x and 905x will not have this issue and we'll plan accordingly when we upgrade.  

Regarding recommendation [2], _use the Specialty Digital modules in an EtherCAT Expansion Chassis (NI 9145)_, I do not have a 9145, but do have 9144.  The disadvantage of the 9411 module in the EtherCAT chassis is that it does not have the Specialty Digital Settings, which makes using a speed sensor or quadrature challenging.  Compare the attached screen shot to the previous screen shot.

A feature request would be to add Specialty Digital Settings support to the 9411 in the 9144 and 9145.  We currently create an FPGA to support this.  
[image]

closed 2019-07-26 by tarmanan