ni/niveristand-scan-engine-ethercat-custom-device #54: Not correct "Add-On Installation Instructions"

ruprotaf, 2019-09-10:
**Describe the bug**
Perository don`t have "Scan Engine" folder.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Veristand: 2019
- LabVIEW: 2019

**Additional context**
If i put folder "Custom Device Source" to "<Public Documents>\National Instruments\NI VeriStand 20xx\Custom Devices" i get error: 
LabVIEW: (Hex 0x7) File not found. The file might be in a different location or deleted. Use the command prompt or the file explorer to verify that the path is correct.

VI Path: C:\Users\Public\Documents\National Instruments\NI VeriStand 2019\Custom Devices\Scan Engine\Scan Engine - Configuration.llb\Scan Engine Initialization VI.vi

buckd, 2019-09-10:
@ruprotaf The source code contained in this repository is not sufficient to use the add-on. The **Add-On Installation Instructions** you refer to assume you have compiled the source code into binaries by building the build specs contained in the `Scan Engine.lvproj` file. For VeriStand 2019, you will need to compile the binaries in LabVIEW 2019.

However, the installation instructions you found are outdated. For VeriStand 2019, there are two ways you can get this support. If you run the VeriStand 2019 installer from NI Package Manager, you will have the option to include the _VeriStand Custom Devices_ add-on. This includes the **Scan Engine and EtherCAT Custom Device**.

Alternatively, you can install the add-on by running the `.nipkg` file from the latest [release](https://github.com/ni/niveristand-scan-engine-ethercat-custom-device/releases). Just make sure to install the correct version support (in this case, 2019).

We will track this issue as a documentation deficiency and work on getting these instructions updated.

Karl-G1, 2020-10-08:
Fixed via documentation in #101

closed 2020-10-08 by Karl-G1