ni/niveristand-scan-engine-ethercat-custom-device #63: Is this Custom Device Async or Inline or Async&Inline?

Kreative24hk, 2019-12-11:
After I added EtherCAT Custom Device on my VeriStand system, It seems that HP Count is increasing now.

I think If the custom device is inline type, It should affect on PCL loop's delay. So, I am curious about whether EtherCAT Custom Device is Async or Inline or Async&Inline.

Is there anyone who can answer this question?

rtzoeller, 2019-12-11:
The custom device can be configured to run either inline or inline/async. This can be controlled using the "Synchronize NI VeriStand to Scan Engine" option on the main page of the custom device.

More specifically, the custom device will run inline if it is set as the Master Custom Device in the target settings (set automatically when checking "Synchronize NI VeriStand to Scan Engine"). If it is not, the custom device will run inline/async, with the decimation being used to set the scan rate.

Kreative24hk, 2019-12-11:
Thanks for your answer!

There is three additional questions.
1. When using this custom device as inline/async, which functions are working at the inline loop? And which functions are working at the async loop?
2. I can find the status channel for async loop’s performance, which is ‘async loop duration’. What kinds of status channel I should be monitoring for inline loop duration?
3. As I know, scan engine in LV has a soft upper limit 1kHz. Does It means that the loop rate of this custom device also meet the rule under 1kHz?

The reason that I ask is... when I use this custom device for Ethercat DIO 320 ch& AIO 96ch in 1kHz, the HP count was increased. So, I just want to know that this custom device can affect the PCL loops latency.

rtzoeller, 2019-12-12:
Hi @Kreative24hk,

1. I would encourage you to look at the source code for the RT engine (`Custom Device Source/RT/Scan Engine RT Driver.vi`), as I believe it will answer your questions better than I will be able to through text on an issue.
2. I don't know that we publish any timing data for the inline execution of the custom device. That executes with the PCL, so for some high-level metrics you could deploy a system definition with only the custom device in it, and monitor performance. Alternatively (but possibly much more work), you could instrument the custom device to log this data.
3. I am not sure of the specifics of Scan Engine's performance. I recommend reaching out to NI Support through the standard support channels (support@ni.com or 1(800)531-5066) for more information there. 

> The reason that I ask is... when I use this custom device for Ethercat DIO 320 ch& AIO 96ch in 1kHz, the HP count was increased. So, I just want to know that this custom device can affect the PCL loops latency.

Any custom device can potentially impact the PCL execution, but it typically depends on the details of the target controller and the other software it is executing. I recommend deploying a system definition with just the custom device and benchmarking your setup for the best answer.

closed 2020-01-06 by Karl-G1