ni/niveristand-scan-engine-ethercat-custom-device #65: Error 537702 while deploying to crio 9047 and crio 9040

hhb-alt, 2019-12-31:
Precondition:
1. Veristand version 2019 & 2019 R2
2. scan-engine-ethercat-custom-device verision 19.2.0.5,19.1.0.5,19.0.0.11
3. Crio 9047 with NI 9403 on slot1(or Crio 9040 with NI 9403 on slot2)

steps:
 1.create a new labview fpga project
 2.add NI9403 into "real time scan resources"
 3.add an empty vi into fpga
 4. compile and get a bitfile
 5.create a new veristand project with crio 9047 or crio 9040
 6.add scan engine custom devcie, leave NI 9403 slot empty. It's running perfect
 7.Select NI9403, it continues report error 537702. configureation mismatch

[image]
[image]
[image]

buckd, 2020-01-02:
@hhb-alt Did you auto-detect the 9203 in System Explorer or did you manually change the module type? The cRIO-904x requires the module mode to be set on a per-module basis, not a per-chassis basis like previous models. There is a known issue with VeriStand for the Controller settings (in your case the `dcc` target), where the **FPGA / Scan interface mode** will not correctly set the mode for per-module targets.

The [19.0.0 release](https://github.com/ni/niveristand-scan-engine-ethercat-custom-device/releases/tag/v19.0.0) of the custom device added support for changing the mode during auto-detect, but if you manually added the module, this code does not execute. In order to make sure the configurations match, you'll need to open **NI MAX** and make sure to configure the module there to use the `Real-Time Scan (IO Variables)` **Programming Mode**.

hhb-alt, 2020-01-03:
@buckd Thank for reply. I've tried both auto-detect and manually change. Auto-detect will remove all configuration and I have to manually add local chasis and modules again. Here are the steps:
1.This was the beginning. I manually added the configuration. It looked like this
[image]
2.Then I chose auto-detect modules
[image]
3.This showed a warning.
[image]
4.Then the local chasis was gone
[image]
5. I clicked the auto-detect again. Local chasis wouldn't be back.
6. Then have to add local chassis again. Add modules,just like step 1.
7. In NI MAX ,module were in "real time scan IO" mode
[image]
8.Once I deployed veristand. Veristand will report 537702 error and stop. Inside NI max, I could observed the 2 modules were change to "labview FPGA" mode automatically.
[image]

buckd, 2020-01-03:
This looks like it stems from the fact that selecting a custom bitfile causes an automatic selection of **FPGA Mode** for the _FPGA / Scan interface mode_ setting on the controller page.

After you select your custom bitfile, go back to your controller page (_dcc_) and you will see **FPGA Mode** has been selected. Try changing the mode back to **Use current** before you deploy. When I made that change, I was able to successfully deploy (you will need to reset the mode to **Real-Time Scan IO** in MAX before doing this). Let me know if this works for you.

Note that any time you change the custom bitfile configuration, this setting will get changed back to **FPGA Mode**, so make sure you set this value after importing your bitfile.

I'm not sure why the auto-detect is failing, but I was able to reproduce that also, but only after a failed deployment. I'll create an issue for that too.

hhb-alt, 2020-01-06:
@buckd Thanks again. Following  your advice, I changed NI 9403 and 9205 to real time scan mode ,   "FPGA/Scan interface mode" to "use current", it's working perfectly.
[image]
I thought it's the "scan engine " changed the working mode of crio 904x. It seems  it is because veristand itself. 
"use current": means module mode will not  be changed
"FPGA mode":means all module  will be changed  to FPGA mode 
"Scan  mode":means  all module will be changed  to scan mode.
I will close this issue . Thanks again.

closed 2020-01-06 by hhb-alt