ni/niveristand-scan-engine-ethercat-custom-device #70: Version in XML doesn't reflect release version.

Sarci1, 2020-01-09:
This can be confusing for the installer, not being able to correlate between GitHub and what is installed.

rtzoeller, 2020-01-20:
Hi @Sarci1,

I agree that this can be a point of confusion, however at this time we are unlikely to change it. The version in XML is primarily used as an internal version number (e.g. for mutation), and isn't really intended to be exposed directly to the user in the normal case.

Using the release version in the XML poses technical challenges. Specifically, it would tie our custom device mutation logic to our release process, which is a coupling that we would like to avoid.

---

Since the custom device is consumed through NIPM, I am proposing removing the version property from the custom device main page. 
[image]

This would encourage customers to check the version in NIPM, which should prevent them from seeing this issue. Does this seem like an acceptable workaround @Sarci1?

Sarci1, 2020-01-20:
I agree with this logic!

PR #71 (2020-01-21): Remove custom device version from main page
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/niveristand-scan-engine-ethercat-custom-device/blob/master/CONTRIBUTING.md).

### What does this Pull Request accomplish?

Removes the version field from the main page of the custom device.

### Why should this Pull Request be merged?

This property displayed the internal mutation version, not the install version. This was a source of confusion and in general not very helpful. Closes #70.

### What testing has been done?

Hand testing of the UI.
closed 2020-01-21 by rtzoeller