ni/niveristand-scan-engine-ethercat-custom-device #86: Can't build the Configuration LLB

andresyie, 2020-06-12:
**Please describe the problem you're facing.**
I'm trying to build this Custom Device but I'm Having problems with Configuration LLB. When I try to build the Configuration specification I get and a error saying some VIs broken during the building. I can trace this broken VIs back to the Modules.lvlibp.

**Describe troubleshooting steps you've taken.**
I tried several versions and also I reinstalled LabVIEW. 

**Additional context**
[image]
[image]

Thanks.

buckd, 2020-06-12:
@andresyie, from the pictures above, your LabVIEW project has not loaded `Modules.lvlibp`, which likely means one of two things. Either the library does not exist on disk where it is expected, or it does exist, but is built for a different LabVIEW version or operating system.

The library is not part of the repo. It must be built from [niveristand-scan-engine-module-libraries](https://github.com/ni/niveristand-scan-engine-module-libraries) or copied from the [latest release](https://github.com/ni/niveristand-scan-engine-module-libraries/releases) from that repo.

Note that you will also need the `NI ECAT Remote IO.llb` from the same repo mentioned above and the [FXP.llb](https://github.com/ni/niveristand-scan-engine-fxp-libraries) library. All of these should be copied to the `Custom Device Source` directory before loading the project.

andresyie, 2020-06-15:
> ote that you will also need the `NI ECAT Remote IO.llb`

Thanks for the fast answer. You are right. I buildt the modules.lvlibp for a RT PXI Target but the configuration.llb is for windows since I´m deploying from a windows 10 PC to a PXI. How can I build a configuration.llb for a windows machine if there is no  modules.lvlibp for windows? Im the project I download I only see for RT, VxWorks, and Linux, 

Thanks.

buckd, 2020-06-15:
> How can I build a configuration.llb for a windows machine if there is no modules.lvlibp for windows? Im the project I download I only see for RT, VxWorks, and Linux,

You're correct. There is not a Windows build of the modules. This is an unfortunate artifact of when this custom device was created long ago. From LabVIEW's perspective, Windows and Pharlap are very similar, and, in this case, the Pharlap binaries can also be loaded on Windows. So, if you copy the Pharlap RT libraries to the `Custom Device Source` directory and then load the project and try to build the `Configuration.llb`, you should be successful.

andresyie, 2020-06-17:
> You're correct. There is not a Windows build of the modules. This is an unfortunate artifact of when this custom device was created long ago. From LabVIEW's perspective, Windows and Pharlap are very [...]

Thanks for the anwer. I builded the RT libraries and put them in the Custom Device Source but now when I try to open the Custom Device Project I get a crash. Any ideas?
[image]

Thanks again for helping :).

Karl-G1, 2020-10-08:
Fixed via documentation in #101

closed 2020-10-08 by Karl-G1