ni/niveristand-scan-engine-ethercat-custom-device #91: SEECD 2020.2 on PharLap cannot run ECAT 9145 in hybrid mode with FPGA bitfile

keprt, 2020-09-09:
**Describe the bug**
ECAT 9145 hybrid mode with FPGA bitfile fails with error 1031 (VI Reference type does not match) on Compare ECAT FPGA Wrapper.vi

**To Reproduce**
Steps to reproduce the behavior:
1. Open the attached TestProject_EtherCAT_FPGA.nivsproj
2. Upload the FPGA bitfile (test has 9474 in slot 8, however the bug could be seen even without model) 
3. Deploy the system definition

**Expected behavior**
Successful deployement 

**Screenshots**

**Desktop (please complete the following information):**
 - OS:  Pharlap
 - Version: 2020

**Additional context**
Reproduction case and screenshots attached

[003 ScanEngineEthercatTest.zip](https://github.com/ni/niveristand-scan-engine-ethercat-custom-device/files/5194316/003.ScanEngineEthercatTest.zip)
[image]
[image]

Karl-G1, 2020-09-09:
### Problem summary
@keprt Thanks for the thorough write-up. I did some digging this morning, and this has been broken for all 20.0+ releases. It stems from @fmussi's change in https://github.com/ni/niveristand-scan-engine-ethercat-custom-device/commit/01bc68dcf98633fcce51d09b7e7edbf84884b431. The 'UDP Data Type.ctl' type def was modified and is reference by the FPGA Data.ctl which is used as an input to the asynchronous call in Compare ECAT FPGA.vi. Because the strictly typed VI ref in the Compare ECAT FPGA Wrapper.vi was not updated in that same submission, the asynchronous call errors out.

### Workaround

[image]

If you look closely, the existing Call by Reference node has a coercion dot on the FPGA in terminal. Creating a new strictly typed VI ref from the current state of the Compare ECAT FPGA.vi and wiring that to the Open VI Reference call fixes the issue.

### Next steps
I will get a fix submitted to the repo shortly, as well as unit tests around the wrapper VIs and a system test for ECAT FPGA I/O. We will likely patch the v20.3.0 release that just went live so we can quickly get this working for VeriStand 2020+ versions.
@keprt if you need a fix quickly, you can use a dev build with the change for testing.

keprt, 2020-09-09:
Hi @Karl-G1  thank you for the quick help. The problem is clear. I will let customer to decide if he will wait on 20.3 patch or he wants to do a dev build until that. 
Unit test would be appreciated.
FYI @czokubik

JiriVejrazka, 2020-09-09:
@Karl-G1 , @keprt : I updated the vi reference, recompiled and I can deploy the project now. I hence do not need to wait for the patch.
Thank you very much!

PR #93 (2020-09-11): Fix EtherCAT FPGA deployment and add related tests
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/niveristand-scan-engine-ethercat-custom-device/blob/master/CONTRIBUTING.md).

### What does this Pull Request accomplish?

- Fix #91 
- Add unit tests for all three 'Wrapper' VIs to ensure async call is not stale
- Add system test for EtherCAT 9145 FPGA mode deployment

### Why should this Pull Request be merged?

#91 broke without the development team knowing due to lack of testing. This fixes the broken feature and adds two different test strategies for ensuring it do [... cut]
closed 2020-09-11 by rtzoeller