ni/niveristand-scan-engine-ethercat-custom-device #104: Scaled Units Supercession in User Interface

tarmanan, 2020-10-19:
**Describe the bug**
Per VeriStand Help Manual, units from Scale should supercede the units associated with the channel.  

https://www.ni.com/documentation/en/veristand/19.2/manual/create-polynomial-scale/

Units supercession carries through to data log file [ie, TDMS], but does not change annotation of units in User Interface. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create Polynomial scale and specify units
2.  Map Scale to channel via Scale Mappings
3.  View Channel in User Interface

**Expected behavior**
User Interface displays channel with units specified in Scale, rather than units from "Channel Settings"

**Screenshots**
[image]
[image]
[image]
[image]
[image]

**Desktop (please complete the following information):**
VeriStand 2019 R3
NI Scan Engine and EtherCAT for VeriStand 2019 v. 20.3.1
cRIO-9068
OS:  Linux

**Additional context**
Add any other context about the problem here.

rtzoeller, 2020-10-19:
Thanks for filing this @tarmanan!

Since this has to do with VeriStand, rather than the Scan Engine EtherCAT custom device specifically, I am going to close this issue.

We've created an internal bug for VeriStand tracking this issue, with ID 1155690.

closed 2020-10-19 by rtzoeller

rtzoeller, 2020-10-30:
@tarmanan I've been unable to reproduce this with VeriStand 2019 R3 - are there any more specific reproducing steps you can share?

rtzoeller, 2020-11-04:
@tarmanan any update on this?

tarmanan, 2020-11-05:
@rtzoeller Thank you for following up.  I attempted to reproduce this again in a new VeriStand 2019 R3 project [Scan Engine EtherCAT custom device v. 20.3.1] today and was unsuccessful as well.