ni/niveristand-scan-engine-ethercat-custom-device #117: 9214 Open Thermocouple Detection OTD Documentation

tarmanan, 2020-11-16:
## If you need immediate assistance, please contact NI Support at support@ni.com or 1(800)531-5066.

**Please describe the problem you're facing.**
Unclear on how to use Open Thermocouple Detection [OTD] using NI-9214 with EtherCAT custom device.  

**Describe the expected behavior**
If a thermocouple fails or is disconnected, I would like the value to be displayed == 300 C.  

**Describe troubleshooting steps you've taken.**
Attempted different OTD compensation values.  If "Enable Open Thermocouple Detection" is unchecked, false values are displayed for thermocouples that are physically disconnected.  If "Enable Open Thermocouple Detection" is checked, the value displayed, for thermocouples physically disconnected, is "0".  I would like this value to be a very high value [ie. 300 C] to trigger a high temperature alarm - ie., if a thermocouple fails or is disconnected, trigger the corresponding high temperature alarm.  

**Additional context**
Add any other context or screenshots about the feature request here.

[image]

[image]

[image]

rtzoeller, 2020-11-16:
@tarmanan unfortunately there is not a [property on the 9214](http://zone.ni.com/reference/en-XX/help/373197L-01/criodevicehelp/crio-9214_rt/) which configures the value returned when an open thermocouple is detected. The custom device doesn't do any runtime processing of the data returned by the Scan Engine API (just initial module configuration), so the values you're seeing are exactly what the underlying API is returning.

There is a workaround involving calculated channels which might address your use case. You could use a conditional calculated channel that checks the 9214 channel, returning 300 if the channel is zero, otherwise returning the 9214 channel value.

[image]

If zero Celsius is an otherwise-valid value for your measurements (i.e. you can't distinguish between an actual zero measurement and an open thermocouple), one option would be to use different units (e.g. Kelvin) and then scale the output. That would allow you to alarm on the original value, which will only be zero in the case of an open thermocouple, while consuming the correctly scaled value in the rest of your application.

tarmanan, 2020-11-16:
@rtzoeller , thank you for this explanation.  So, if "0" is entered for for all of the compensation values, I should see 0 C when an open thermocouple is detected?  

I think the issue I am observing is all channels after the open thermocouple are "0" or at Max Value.  See Slot 6 TC 6, which is open, goes to a a very high value.  Slot 6, TC 7 and higher go to 0 even though Slot 6, TC 12 is plugged in a sending a valid signal.  

In the second screen shot, I have moved the thermocouple from Slot 6, TC 12 to Slot 7, TC 0 and it functions properly.

[image]

[image]

Karl-G1, 2020-11-17:
@tarmanan I did some further troubleshooting today to see if this behavior is caused from the custom device or from the Scan Engine itself. I have a 9214 module in a cRIO-9045, and although I have no thermocouples attached, I believe I'm seeing consistent behavior.

One thing to clear up first: the Open-Thermocouple Detection (OTD) compensation does not impact the reported value directly. Rather, it is to compensate for the small measurement error introduced when OTD is enabled. You can read more here: https://www.ni.com/en-us/support/documentation/supplemental/10/overview-of-open-thermocouple-detection.html

The behavior we are seeing is a by-product of how the Scan Engine treats the values when the OTD is triggered. Effectively, when reading back multiple values from the driver in the typical fashion, the call returns an error and no values. When reading back the container uninitialized, it returns one value and the rest are zeroes, which is arguably better and the workaround employed by the custom device. I verified that there is no real alternative to this behavior by testing the same calls in LabVIEW:

[image]

I have a feeling that internally, the call is just executing a For loop to read the published values, and an error is probably propagated causing the rest of the reads (even for connected thermocouple channels) to fail. Unfortunately, there is nothing we can workaround in the custom device to fix this, and a driver fix is unlikely to happen anytime soon.

If OTD is required for this application, the best workaround is probably to move the 9214 value reads to the FPGA. On FPGA, we can read out the Open/Closed status of each channel in a Boolean array, and then you can report whatever value you want when it is open.

This is fairly cumbersome to configure the first time, but the logic to report back values is straightforward. You just need to configure a UDV for each channel, then load the bitfile into the custom device. When doing this, I can see that all my channels are open, the raw value returned is 0, and my open value is 300.

[image]

[image]

[image]

I did not look too much into scaling the thermocouple values, as this would take a while to get executing on the FPGA side. There is [an example](https://zone.ni.com/reference/en-XX/help/370984T-01/criodevicehelp/9214_converting/) that has the logic to use the fixed-point value and the CJC for each channel to calculate the temperature, but this is provided as a host-side VI. You would likely want to do the calculation on FPGA to simplify integration with VeriStand.

tarmanan, 2020-11-17:
Thank you, @Karl-G1 .  This explanation is helpful and helps me better understand how the Custom Device interacts with OTD as well as gives me some options for future projects.

I should have mentioned this earlier, but I am using VeriStand 2019 R3, cRIO-9045, and the 9214 are in a 9144 EtherCAT chassis.

closed 2020-11-17 by tarmanan