ni/niveristand-scan-engine-ethercat-custom-device #168: Variable Refnum Invalid Error When deploy EtherCAT Custom device at second time.

trionix-hiro, 2021-08-04:
## If you need immediate assistance, please contact NI Support at support@ni.com or 1(800)531-5066.

**Please describe the problem you're facing.**
Veristand return error below, when deploy EtherCAT Custom device at second time.
It dose not return error at first time after RT reboot.
But it return error, after un-deploy  system then deploy again.

[image]

* sorry for Japanese error massage, it describe that "Variable refnum is invalid. variable is deleted after opening reference or valid reference is not opened"

[Environment]
Veristand 2020 R5
Scan engine & Ethercat custom device (Veristand 2020 R5 shipping version) 
CompactRIO 9055
Third-Party EtherCAT Slave (Comemso Battery Cell Simulator)

**Describe the expected behavior**
it should not return error when re-deploy the system.
Currently, I have to always re-boot system to deploy it again.

**Describe troubleshooting steps you've taken.**
First, I made simple LabVIEW program to test third party EtherCAT slave, and it work properly in LabVIEW environment.

**Additional context**

Karl-G1, 2021-08-04:
This sounds like an issue with how the custom device or the EtherCAT slave are handling state transitions. As a quick test to make sure it works from LabVIEW in the same manner, make sure to transition the Scan Engine between Active and Configuration modes in the project. Otherwise, just deploying to the system from LabVIEW without changing the state is not a valid comparison.

Either way though, this is not the best support forum for this issue. I recommend creating a support case via the email or phone numbers above. NI technical support can help with more detailed debugging and next steps.

trionix-hiro, 2021-08-06:
thanks for comments.
I tested with LabVIEW following procedure and EtherCAT slave works properly.
1. Set Scan engine into configuration mode at utility menu.
2. then Set Scan engine into active mode and deploy RT VI.  
3. Stop RT VI and set Scan engine into Configuration mode.
4. Again Set scan engine into active mode then, re-deploy RT VI.

Do you think above procedures are same as Veristand do when deploy & undeploy?

I will also try to contact NI tech support.

best regard.

Karl-G1, 2021-10-14:
Closing for inactivity.

closed 2021-10-14 by Karl-G1