ni/niveristand-scan-engine-ethercat-custom-device #174: Error trying to access modules

lsmarin75, 2021-09-28:
## If you need immediate assistance, please contact NI Support at support@ni.com or 1(800)531-5066.

**Please describe the problem you're facing.**
I have this configuration :
[image]
I am using 2 EtherCat chassis one with all the digital modules and the second one with the analog I/O, I have the following problems:

1- With the first chassis, when I am trying to deply I am getting the error: Error 537702 occurred at Project Window.lvlib:Project Window.vi >> Project Window.lvlib:Command Loop.vi >> Project Window.lvlib:Connect to System.vi.

My question here is: Is the configuration shared in the picture supported by the CD, or should I remove some digital modules?
2- I need the NI 9238 analog input module, which is not supported on scan interface, so, I have created a custom bitfile, in hybrid mode, but when I try to deploy I get the error 7 file not found but I am not sure about what file :
[image]

Possible reasons:
There was a mismatch between the modules configured in the system definition and what was detected on the target system.

Possible reasons:
There was a mismatch between the modules configured in the system definition and what was detected on the target system.

**Describe the expected behavior**
A clear and concise description of what you expected to happen.

**Describe troubleshooting steps you've taken.**
I have remove my modules and place them again, I have formatted my PXI controller and then I reinstalled the SW needed 

**Additional context**

lsmarin75, 2021-09-28:
update:

I decided to reinstall Veristand to see it helps:

1.- I tested just with one chassis with all the digital moodules and the system deploy with no errors.
2.- I connected the second one chassis with the analog modules without the custom bitfile the system does deploy with no errors but with the custom bit file I get error:

 "Targets/controller/customDevices/ScanEngineand EtherCAT"Error 7 occurred at project window.lvlib
LabVIEW hex 0x7 file not found.
When a try to open de InitializationVI.VI I got error :LabVIEW hex 0x7 file not found. LabVIEW load error code 3 Could not load front panel.
I am using Veristand 2020 and LabVIEW 2020

I will appreciate your comments !
Best regards

rtzoeller, 2021-09-28:
Hi @lsmarin75,

I think this issue is going to be best served through NI's traditional support channels rather than via a GitHub issue. Can you reach out to NI support at support@ni.com or 1(800)531-5066?

I'm also having some trouble viewing the embedded images in your original post, as they are too small to read on my machine. You may want to provide new ones when contacting NI support.

Best,
Ryan

Karl-G1, 2021-10-14:
It looks like this was solved by the workaround in #175. I'll close this issue out as a duplicate.

closed 2021-10-14 by Karl-G1