ni/niveristand-scan-engine-ethercat-custom-device #226: Custom Device User Help Documentation

tarmanan, 2023-02-28:
**Is your feature request related to a problem? Please describe.**
Help documentation for the Scan Engine and EtherCAT custom device requires external sources such as LabVIEW, web help documents

**Describe the solution you'd like**
Most of our VeriStand / Scan Engine and EtherCAT Custom Device installations do not have LabVIEW installed or internet access. 
Our team has very little LabVIEW experience.  For example, to find information on configuration of specialty digital settings, it requires using a PC with LabVIEW and/or internet access to understand configuration settings

**Describe alternatives you've considered**
PC with LabVIEW and/or internet access

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

[image]

Karl-G1, 2023-03-01:
@tarmanan 
Would adding a simple button to the Slot page that launches the help be a good enough solution? We could include the CompactRIO and C Series CHM files in the Scan Engine Custom Device directory, then link to it from the page fairly easily for something like this:
[image]

I could see a solution where we link to the specific module's help page once it is selected (requiring a change in the [modules libraries](https://github.com/ni/niveristand-scan-engine-module-libraries) which serves the sub-panels with the configuration options). However, that would take a lot more time to implement.

tarmanan, 2023-03-02:
@Karl-G1 , this would be an excellent solution!

Karl-G1, 2023-03-15:
This fix will be included in the next release. Closing issue.

closed 2023-03-15 by Karl-G1